tjhowse/modbus4mqtt #12: [v0.3.0] TypeError on run

bacco007, 2020-10-11:
installed 0.3.0 and got the following error

```
pi@TBSH02:~ $ /home/pi/.local/bin/modbus4mqtt --hostname 192.168.1.63 --username mqtt --password mqtt --config config.yaml
Traceback (most recent call last):
  File "/home/pi/.local/bin/modbus4mqtt", line 10, in <module>
    sys.exit(main())
  File "/home/pi/.local/lib/python3.7/site-packages/click/core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/click/core.py", line 782, in main
    rv = self.invoke(ctx)
  File "/home/pi/.local/lib/python3.7/site-packages/click/core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.local/lib/python3.7/site-packages/click/core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.local/lib/python3.7/site-packages/modbus4mqtt/modbus4mqtt.py", line 169, in main
    i.connect()
  File "/home/pi/.local/lib/python3.7/site-packages/modbus4mqtt/modbus4mqtt.py", line 29, in connect
    self.connect_modbus()
  File "/home/pi/.local/lib/python3.7/site-packages/modbus4mqtt/modbus4mqtt.py", line 38, in connect_modbus
    while self._mb.connect():
  File "/home/pi/.local/lib/python3.7/site-packages/modbus4mqtt/modbus_interface.py", line 37, in connect
    RetryOnEmpty=True, retries=1)
TypeError: 'module' object is not callable
```

tjhowse, 2020-10-11:
Thanks for the report!

Can you please provide a copy of your config.yaml?

bacco007, 2020-10-11:
I used https://github.com/tjhowse/modbus4mqtt/blob/master/modbus4mqtt/Sungrow_SH5k_20.yaml and changed the ip address to the address of my inverter

tjhowse, 2020-10-11:
Cheers! I'm investigating now. Unfortunately my inverter is running the older, pre-encryption version of the firmware so I'm unable to test it easily locally. Stand by!

tjhowse, 2020-10-11:
OK. I believe I've found the issue. I've pushed up version 0.3.1. It would be great if you could give that a test and see if everything's working.

bacco007, 2020-10-11:
That appears to have worked - just need to work out why it wont read the modbus register :)

tjhowse, 2020-10-11:
Great, let me know if I can help troubleshoot your issue.

closed 2020-10-11 by tjhowse