tjhowse/modbus4mqtt #13: ERROR 'ExceptionResponse' object has no attribute 'registers

plague-doctor, 2020-10-28:
I try to test your script (it would be soooo useful to me).

My Inverter: SG5K-D
With latest firmware: M_WIFI_RAK475_V31_V01_Q

modbus4mqtt --hostname mqttserver --port 1883 --username username --password password --mqtt_topic_prefix test --config ./modbus4mqtt/Sungrow_SH5k_20.yaml

```
2020-10-28 13:50:29 INFO     Connected to MQTT.
Subscribed to test/no_export/partial/limit/set
Subscribed to test/forced_charge/mode/set
Subscribed to test/forced_charge/weekdays/set
Subscribed to test/forced_charge/period_1/start_hours/set
Subscribed to test/forced_charge/period_1/start_minutes/set
Subscribed to test/forced_charge/period_1/end_hours/set
Subscribed to test/forced_charge/period_1/end_minutes/set
Subscribed to test/forced_charge/period_1/target_soc/set
Subscribed to test/forced_charge/period_2/start_hours/set
Subscribed to test/forced_charge/period_2/start_minutes/set
Subscribed to test/forced_charge/period_2/end_hours/set
Subscribed to test/forced_charge/period_2/end_minutes/set
Subscribed to test/forced_charge/period_2/target_soc/set
Subscribed to test/bat_usage_time/weekday_usage/end_time_1_hours/set
Subscribed to test/bat_usage_time/weekday_usage/end_time_2_hours/set
Subscribed to test/bat_usage_time/weekend_usage/set
2020-10-28 13:50:30 ERROR    Failed to poll modbus device, attempting to reconnect: 'ExceptionResponse' object has no attribute 'registers'
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus4mqtt.py", line 71, in poll
    self._mb.poll()
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 58, in poll
    values = self._scan_value_range(table, group, DEFAULT_SCAN_BATCHING)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 128, in _scan_value_range
    return result.registers
AttributeError: 'ExceptionResponse' object has no attribute 'registers'
2020-10-28 13:50:35 ERROR    Failed to poll modbus device, attempting to reconnect: 'ExceptionResponse' object has no attribute 'registers'
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus4mqtt.py", line 71, in poll
    self._mb.poll()
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 58, in poll
    values = self._scan_value_range(table, group, DEFAULT_SCAN_BATCHING)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 128, in _scan_value_range
    return result.registers
AttributeError: 'ExceptionResponse' object has no attribute 'registers'
2020-10-28 13:50:42 ERROR    Failed to read holding table registers from 13000 to 13100
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 58, in poll
    values = self._scan_value_range(table, group, DEFAULT_SCAN_BATCHING)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 127, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read holding table registers from 13000 to 13100
2020-10-28 13:50:42 ERROR    Unable to decode response Modbus Error: Unknown response 0
2020-10-28 13:50:42 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/transaction.py", line 190, in execute
    self.client.framer.processIncomingPacket(response,
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2020-10-28 13:50:42 ERROR    Failed to read holding table registers from 13100 to 13200
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 58, in poll
    values = self._scan_value_range(table, group, DEFAULT_SCAN_BATCHING)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 127, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read holding table registers from 13100 to 13200
2020-10-28 13:50:42 WARNING  Couldn't get value from register 13073 in table holding
2020-10-28 13:50:42 WARNING  Couldn't get value from register 13139 in table holding
2020-10-28 13:50:42 WARNING  Couldn't get value from register 13140 in table holding
[... 67 lines cut ...]
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 58, in poll
    values = self._scan_value_range(table, group, DEFAULT_SCAN_BATCHING)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 127, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))

[...]
```
and so on, and so on...

cat modbus4mqtt/Sungrow_SH5k_20.yaml

```
ip: 172.16.50.116
port: 502
update_rate: 5
address_offset: 0
variant: sungrow
registers:
  - pub_topic: "no_export/partial/limit"
    set_topic: "no_export/partial/limit/set"
    address: 13073
[...]
# The rest of lines are the same as provided in repo
```

tjhowse, 2020-10-28:
Hey! Thanks for the thorough report. It really helps. Sorry you're having trouble.

It looks as though the modbus interface isn't happy talking to your inverter. There can be a few reasons for this. Are you able to verify that the inverter is listening on port 502 at that IP and there's nothing blocking the connection? You could do this by doing a port scan with `nmap`.

This test shows port 502 is open for connections on my inverter, and 501 is not.
```
tjhowse@home:~$ nmap 192.168.1.89 -p 502

Starting Nmap 7.40 ( https://nmap.org ) at 2020-10-28 14:02 AEST
Nmap scan report for 192.168.1.89
Host is up (0.0027s latency).
PORT    STATE SERVICE
502/tcp open  mbap

Nmap done: 1 IP address (1 host up) scanned in 0.06 seconds
tjhowse@home:~$ nmap 192.168.1.89 -p 501

Starting Nmap 7.40 ( https://nmap.org ) at 2020-10-28 14:02 AEST
Nmap scan report for 192.168.1.89
Host is up (0.0031s latency).
PORT    STATE  SERVICE
501/tcp closed stmf

Nmap done: 1 IP address (1 host up) scanned in 0.06 seconds
```

Or if you don't have `nmap` you could try `netcat`:

```
tjhowse@home:~$ netcat 192.168.1.89 501
(UNKNOWN) [192.168.1.89] 501 (?) : Connection refused
tjhowse@home:~$ netcat 192.168.1.89 502
^C
```
(Press ctrl-c to terminate netcat if a connection is successful)

Unfortunately Sungrow recently encrypted the modbus communication with their newer firmwares. v0.3.1 of modbus4mqtt supports the `variant:` config field which, when set to `sungrow`, decrypts the modbus comms using https://github.com/rpvelloso/Sungrow-Modbus

Unfortunately I don't have an inverter with the encrypted modbus, so I'm unable to test it myself. 

Could you also please confirm the versions of `modbus4mqtt` and `SungrowModbusTcpClient` with:

```
pip3 freeze | egrep "SungrowModbusTcpClient|modbus4mqtt"
```

Cheers!

plague-doctor, 2020-10-28:
Sure thing!
I have already verified the ports are opened, but here you go:
```
$ nmap -p- 172.16.50.116

Starting Nmap 7.80 ( https://nmap.org ) at 2020-10-28 16:37 AEDT
Nmap scan report for 172.16.50.116
Host is up (0.040s latency).
Not shown: 65533 closed ports
PORT    STATE SERVICE
80/tcp  open  http
502/tcp open  mbap
```
and versions:
```
modbus4mqtt==0.3.1
SungrowModbusTcpClient==0.1.4
```
I will be happy to assist with any test you can come up with :-)

tjhowse, 2020-10-28:
Cheers! Tonight I'll put together a quick troubleshooting script that provides some additional information about what's going on inside the modbus connection attempts.

It's possible that Sungrow have changed their crypto again, or different models of inverters have different private keys. Investigation needed.

tjhowse, 2020-10-29:
Could you please grab a network traffic log during a connection attempt to the inverter? 

`sudo tcpdump -i eth0 -w ./log.pcap host 172.16.50.116`

Leave that running in the background then launch `modbus4mqtt` as you normally would.

Should do the trick. You'll probably need to change the ethernet interface name from `eth0` to whichever your system is using to communicate with the inverter. If you could attach that log file to this issue that would be grand.

Warning: The log will likely contain the serial number of your inverter. Feel free to email it directly to me if you want to keep it more private.

plague-doctor, 2020-10-29:
Log in your email.

tjhowse, 2020-10-30:
Brilliant. I can see that the interface is correctly querying for your inverter's key on startup, and that your inverter is responding to that request, but then it goes off the rails. Do you mind if I share this packet log with @rpvelloso, the author of SungrowModbusTcpClient?

tjhowse, 2020-10-30:
OK. I think I might have cracked it. SungrowModbusTcpClient overrides the `connect()` method of ModbusTCPClient. I don't explicitly call `connect()` in my code, it's called when the interface first attempts a `read_holding_registers`. However when connecting to an encrypted server the first connection attempt returns a failure, after it obtains and stores the inverter's public key so a subsequent connect can succeed. However this key erased because when the connection fails I dump and re-create the `SungrowModbusTcpClient` object.

If this is correct then I think there are two options:
1) Modify `SungrowModbusTcpClient` to attempt a reconnection after its `_getkey()` call:
```python
    def connect(self):
        self.close()
        result = ModbusTcpClient.connect(self)
        if not result:
            self._restore()
        else:
            self._getkey()
            self.close()
            result = ModbusTcpClient.connect(self)
        return result
```
This would mean that `SungrowModbusTcpClient` implements connect logic that more closely matches `ModbusTcpClient`'s behaviour.

or 2) Modify `modbus4mqtt` to explicitly call `connect()` twice after initialising a `SungrowModbusTcpClient` object.

@rpvelloso What do you reckon? I'm happy to do the work for either approach.

Cheers,
tjhowse.

plague-doctor, 2020-10-30:
> [...] Do you mind if I share this packet log with @rpvelloso, the author of SungrowModbusTcpClient?

As long as you do not make it public, go for it :-) And again, I am happy to run more tests if you need.

rpvelloso, 2020-10-30:
Good job @tjhowse I think this is going to work o/
merge PR and updated pypi to version 0.1.5: https://pypi.org/project/SungrowModbusTcpClient/0.1.5/

rpvelloso, 2020-10-30:
I haven't looked at the pcap file yet

rpvelloso, 2020-10-30:
installed version 0.1.5 on my setup, it's working.

plague-doctor, 2020-10-30:
Nope... Still not good.

```
 pip3 freeze | egrep "SungrowModbusTcpClient|modbus4mqtt"

modbus4mqtt==0.3.1
SungrowModbusTcpClient==0.1.5
```
still throwing:
```
2020-10-31 08:12:10 ERROR    Failed to poll modbus device, attempting to reconnect: 'ExceptionResponse' object has no attribute 'registers'
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus4mqtt.py", line 71, in poll
    self._mb.poll()
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 58, in poll
    values = self._scan_value_range(table, group, DEFAULT_SCAN_BATCHING)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 128, in _scan_value_range
    return result.registers
AttributeError: 'ExceptionResponse' object has no attribute 'registers'
```

tjhowse, 2020-10-31:
Hmm. Strange. I threw together a very basic test script here: https://github.com/tjhowse/modbus4mqtt/blob/Basic_test/troubleshooting/basic_test.py

Change the IP on line 6 to the IP of your inverter, `chmod+x basic_test.py` and then `./basic_test.py`. You should get something like:

```
ReadRegisterResponse (10)
[10, 31, 14, 0, 54, 0, 0, 0, 0, 0]
ReadRegisterResponse (10)
[50, 0, 201, 22771, 1, 7413, 0, 517, 0, 0]
```

I fear that perhaps Sungrow have changed their encryption again.

plague-doctor, 2020-10-31:
ReadRegisterResponse (10)
[10, 31, 14, 10, 41, 207, 170, 1000, 255, 1000]
ReadRegisterResponse (10)
[50, 0, 102, 165, 0, 125, 0, 406, 65535, 65535]

tjhowse, 2020-10-31:
Interesting. So that confirms that at least those smaller reads starting at 5000 work, which is great. 

I've pushed up a change to the test that reads a bigger range of registers with both basline `SungrowModbusTcpClient` and the `modbus_interface` wrapper. You should get something like this:

```
Basline test. Read 10 register starting at 5000.
ReadRegisterResponse (10)
[10, 31, 14, 26, 49, 0, 0, 0, 0, 0]
ReadRegisterResponse (10)
[50, 0, 209, 22779, 1, 7413, 0, 528, 0, 0]
Basline test. Read 100 register starting at 5000.
ReadRegisterResponse (100)
[10, 31, 14, 26, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 999, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
ReadRegisterResponse (100)
[50, 0, 209, 22779, 1, 7413, 0, 528, 0, 0, 3101, 11, 3452, 11, 0, 0, 619, 0, 2451, 0, 0, 27, 0, 0, 0, 0, 0, 0, 0, 0, 618, 0, 65532, 65535, 999, 500, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 97, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 618, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
modbus_interface test. Read 10 registers starting at 5000.
[10, 31, 14, 26, 50, 0, 0, 0, 0, 0]
[50, 0, 209, 22779, 1, 7413, 0, 528, 0, 0]
modbus_interface test. Read 100 registers starting at 5000.
[10, 31, 14, 26, 50, 0, 0, 0, 0, 0, 0, 0, 0, 0, 999, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
[50, 0, 209, 22779, 1, 7413, 0, 528, 0, 0, 3148, 11, 3468, 10, 0, 0, 628, 0, 2448, 0, 0, 25, 0, 0, 0, 0, 0, 0, 0, 0, 628, 0, 65528, 65535, 1000, 500, 0, 0,0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 97, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 629, 0, 0,0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
```

plague-doctor, 2020-10-31:
```
Basline test. Read 10 register starting at 5000.
ReadRegisterResponse (10)
[10, 31, 14, 40, 28, 207, 170, 1000, 255, 1000]
ReadRegisterResponse (10)
[50, 0, 106, 166, 0, 125, 0, 407, 65535, 65535]
Basline test. Read 100 register starting at 5000.
ReadRegisterResponse (100)
[10, 31, 14, 40, 29, 207, 170, 1000, 255, 1000, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 1000, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 85, 0, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 500, 1000, 1000, 900, 2288, 2200, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 200, 500, 1000, 950, 1050, 1000, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 900, 1152, 957, 1087, 0, 300, 300, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 920, 900, 1080]
ReadRegisterResponse (100)
[50, 0, 106, 166, 0, 125, 0, 407, 65535, 65535, 3328, 12, 3303, 11, 65535, 65535, 791, 0, 2420, 65535, 65535, 32, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 787, 0, 82, 0, 993, 501, 65535, 0, 2020, 10, 26, 7, 49, 0, 0, 65535, 65535, 100, 15, 0, 0, 0, 0, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 32767, 32767, 32767, 32767, 32767, 32767, 32767, 32767, 1142, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 1, 0, 65535, 32767, 65535, 32767, 65535, 32767, 65535, 32767, 65535, 32767, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535]
modbus_interface test. Read 10 registers starting at 5000.
[10, 31, 14, 40, 30, 207, 170, 1000, 255, 1000]
[50, 0, 106, 166, 0, 125, 0, 407, 65535, 65535]
modbus_interface test. Read 100 registers starting at 5000.
[10, 31, 14, 40, 31, 207, 170, 1000, 255, 1000, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 1000, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 85, 0, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 500, 1000, 1000, 900, 2288, 2200, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 200, 500, 1000, 950, 1050, 1000, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 900, 1152, 957, 1087, 0, 300, 300, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 920, 900, 1080]
[50, 0, 106, 166, 0, 125, 0, 407, 65535, 65535, 3317, 12, 3293, 11, 65535, 65535, 790, 0, 2425, 65535, 65535, 32, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 787, 0, 81, 0, 991, 501, 65535, 0, 2020, 10, 26, 7, 49, 0, 0, 65535, 65535, 100, 15, 0, 0, 0, 0, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 32767, 32767, 32767, 32767, 32767, 32767, 32767, 32767, 1142, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 1, 0, 65535, 32767, 65535, 32767, 65535, 32767, 65535, 32767, 65535, 32767, 65535, 65535, 65535, 65535, 65535, 65535, 65535, 65535]
```

tjhowse, 2020-10-31:
This is a sneaky bug! I've pushed up a minimal config file here: https://github.com/tjhowse/modbus4mqtt/blob/Basic_test/troubleshooting/test.yaml

Subscribe to the test topic with `mosquitto_sub -h mqttserver -t 'test/#' -v -u username -P password`

Then start `modbus4mqtt` with that config file instead of `Sungrow_SH5k_20.yaml`

You should see something like this:

```
test/modbus4mqtt hi
test/seconds 47
test/seconds 52
```

Note that the occasional modbus error thrown by `modbus4mqtt` is not an indicator of a total failure. The modbus interface on these inverters isn't particularly robust, so the occasional message gets garbled. 98% of them should be fine though.

Thanks for all your help pinning down this bug. I appreciate it! I really need to scrounge up another of these inverters for testing purposes. The only unit I have to test with runs my house, so I'm leery of breaking it with a firmware update 😟

plague-doctor, 2020-10-31:
test/modbus4mqtt hi
test/seconds 40
test/seconds 45
test/seconds 51
test/seconds 57
test/seconds 2
test/seconds 8
test/seconds 14

No problem. We need to fix this :-)

tjhowse, 2020-10-31:
Righto. It looks like the SG5K-D rejects reads on some of registers that the SH5K-20 accepts. I'm afraid you're going to have to gradually add registers from `Sungrow_SH5k_20.yaml` to `test.yaml` until you find the ones that your inverter doesn't support.

Unfortunately Sungrow don't publish the register map, so it's going to be a process of elimination. My system was to add wide ranges of registers (13000-13300) to a config file without specific names, like this:

```yaml
- pub_topic: "13100"
    address: 13100
  - pub_topic: "13101"
    address: 13101
  - pub_topic: "13102"
    address: 13102
  - pub_topic: "13103"
    address: 13103
  - pub_topic: "13104"
    address: 13104
  - pub_topic: "13105"
    address: 13105
  - pub_topic: "13106"
    address: 13106
  - pub_topic: "13107"
    address: 13107
  - pub_topic: "13108"
    address: 13108
  - pub_topic: "13109"
    address: 13109
  - pub_topic: "13110"
    address: 13110
```

Then I'd subscribe to `test/#` and change a setting via the LCD GUI on the front of the inverter. If the setting I changed was stored in the scanned range the value would publish to MQTT and I'd then add that register to `Sungrow_SH5k_20.yaml` using the exact text shown on the menu.

For settings that I couldn't change via the LCD, such as `no_export/partial/limit`, it was slightly trickier. The default value of that register was `5000`, so I looked through all the values published and found one with that same value. I changed the value via modbus and confirmed that the setting changed value on the LCD.

Once you've got a set of known good registers that your inverter is happy to accept I'd love to add them to add that config file to this repo so people with your inverter have an easier time in the future.

There's a starting point for register addresses and descriptions here: https://github.com/thomasfa18/solar-sungrow/blob/master/modbus-sungrow.py

plague-doctor, 2020-10-31:
The above list generates same error.
Can't you loop through registers and spit out the ones which returns a value? passing on the errored ones?

rpvelloso, 2020-10-31:
So, the problem is with register addresses??? I have this document I got from sungrow itself. According to sungrow this are the addresses for residential models (SG4K, SG5K, etc). I own a SG4K and I'm using the following registers without problem:
```sensor:
  - platform: modbus
    scan_interval: 60
    registers:
      - name: fault_code
        hub: sungrow_sg4k
        slave: 1
        count: 1
        register: 5044
        register_type: input
      - name: total_dc_power
        hub: sungrow_sg4k
        slave: 1
        register: 5016
        count: 2
        register_type: input
        reverse_order: true
        unit_of_measurement: W
      - name: daily_power_yields
        hub: sungrow_sg4k
        slave: 1
        count: 1
        register: 5002
        register_type: input
        scale: 0.1
        precision: 2
        unit_of_measurement: kWh      
      - name: internal_temperature
        hub: sungrow_sg4k
        slave: 1
        count: 1
        register: 5007
        register_type: input
        scale: 0.1
        precision: 2
        unit_of_measurement: ºC
      - name: daily_running_time
        hub: sungrow_sg4k
        slave: 1
        count: 1
        register: 5112
        register_type: input
        scale: 1
        unit_of_measurement: min
      - name: total_active_power
        hub: sungrow_sg4k
        slave: 1
        count: 2
        register: 5030
        reverse_order: true
        register_type: input
        unit_of_measurement: W
      - name: grid_frequency
        hub: sungrow_sg4k
        slave: 1
        count: 1
        register: 5035
        register_type: input
        scale: 0.1
        precision: 2
        unit_of_measurement: Hz
      - name: phase_voltage
        hub: sungrow_sg4k
        slave: 1
        count: 1
        register_type: input
        register: 5018
        scale: 0.1
        precision: 2
        unit_of_measurement: V
      - name: phase_current
        hub: sungrow_sg4k
        slave: 1
        count: 1
        register_type: input
        register: 5021
        scale: 0.1
        precision: 2
        unit_of_measurement: A
      - name: total_power_yields
        hub: sungrow_sg4k
        slave: 1
        reverse_order: true
        count: 2
        scale: 0.1
        precision: 2
        register: 5143
        register_type: input
        unit_of_measurement: kWh
      - name: device_status
        hub: sungrow_sg4k
        slave: 1
        count: 1
        register_type: input
        register: 5037
      - name: mppt1_voltage
        hub: sungrow_sg4k
        slave: 1
        count: 1
        register_type: input
        unit_of_measurement: V
        register: 5010
        scale: 0.1
        precision: 2
      - name: mppt1_current
        hub: sungrow_sg4k
        slave: 1
        count: 1
        register_type: input
        unit_of_measurement: A
        register: 5011
        scale: 0.1
        precision: 2
      - name: mppt2_voltage
        hub: sungrow_sg4k
        slave: 1
        count: 1
        register_type: input
        unit_of_measurement: V
        register: 5012
        scale: 0.1
        precision: 2
      - name: mppt2_current
        hub: sungrow_sg4k
        slave: 1
        count: 1
        register_type: input
        unit_of_measurement: A
        register: 5013
        scale: 0.1
        precision: 2
      - name: power_factor
        hub: sungrow_sg4k
        slave: 1
        count: 1      
        register_type: input
        register: 5034
        scale: 0.001
        precision: 4
```

[Modbus RS485 RTU Protocol.pdf](https://github.com/tjhowse/modbus4mqtt/files/5469426/Modbus.RS485.RTU.Protocol.pdf)

plague-doctor, 2020-10-31:
how can I use this with `modbus4mqtt`?

plague-doctor, 2020-10-31:
Ok, I have modified the test.yaml a bit to reflect above registers:
```
ip: 172.16.50.116
# ip: 192.168.1.89
port: 502
update_rate: 5
address_offset: 0
variant: sungrow
registers:
  - pub_topic: "seconds"
    address: 5004
    pub_only_on_change: false
    table: 'holding'
  - pub_topic: "5016"
    address: 5016
  - pub_topic: "5002"
    address: 5002
  - pub_topic: "5007"
    address: 5007
  - pub_topic: "5012"
    address: 5112
  - pub_topic: "5030"
    address: 5030
  - pub_topic: "5035"
    address: 5035
  - pub_topic: "5018"
    address: 5018
  - pub_topic: "5021"
    address: 5021
  - pub_topic: "5143"
    address: 5143
  - pub_topic: "5037"
    address: 5037
  - pub_topic: "5010"
    address: 5010
  - pub_topic: "5011"
    address: 5011
  - pub_topic: "5012"
    address: 5012
  - pub_topic: "5013"
    address: 5013
  - pub_topic: "5043"
    address: 5043
```
Now I get some values:
```
test/modbus4mqtt hi
test/seconds 7
test/5016 65535
test/5002 7
test/5007 1000
test/5012 44975
test/5030 65535
test/5035 85
test/5018 1000
test/5021 65535
test/5143 16
test/5037 65535
test/5010 65535
test/5011 65535
test/5012 65535
test/5013 65535
test/5043 65535
test/seconds 13
test/seconds 19
test/seconds 26
test/seconds 32
test/seconds 38
test/seconds 44
[...]
```
some of them are ok:
```
test/5002 7
```
Same on the **iSolarCloud: 0.7** but other are rubbish.
The modbus4mqtt throws errors as well, but keeps publishing **test/seconds** (I presume the other values are not changing):
```
2020-11-01 08:11:08 INFO     Connected to MQTT.
2020-11-01 08:12:30 ERROR    Failed to read holding table registers from 5000 to 5100
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 58, in poll
    values = self._scan_value_range(table, group, DEFAULT_SCAN_BATCHING)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 127, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read holding table registers from 5000 to 5100
2020-11-01 08:12:30 ERROR    Unable to decode response Modbus Error: Unknown response 0
2020-11-01 08:12:30 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/transaction.py", line 190, in execute
    self.client.framer.processIncomingPacket(response,
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2020-11-01 08:12:30 ERROR    Failed to read holding table registers from 5100 to 5200
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 58, in poll
    values = self._scan_value_range(table, group, DEFAULT_SCAN_BATCHING)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 127, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read holding table registers from 5100 to 5200
2020-11-01 08:12:36 ERROR    Failed to read holding table registers from 5000 to 5100
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 58, in poll
    values = self._scan_value_range(table, group, DEFAULT_SCAN_BATCHING)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 127, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read holding table registers from 5000 to 5100
2020-11-01 08:12:37 ERROR    Unable to decode response Modbus Error: Unknown response 0
2020-11-01 08:12:37 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/transaction.py", line 190, in execute
    self.client.framer.processIncomingPacket(response,
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2020-11-01 08:12:37 ERROR    Failed to read holding table registers from 5100 to 5200
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 58, in poll
    values = self._scan_value_range(table, group, DEFAULT_SCAN_BATCHING)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 127, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read holding table registers from 5100 to 5200
```

plague-doctor, 2020-10-31:
And some more updates: 
I kept the script running for a while, but only the `test/seconds` keeps changing, even if the `test/5002 7` has changed already. No new value of 5002 register is published.

tjhowse, 2020-10-31:
So I think the above PR might be helpful in making it easier to scan novel devices. In this case we could set it to 1 and each register would be scanned individually. This means one unsupported register won't break an entire read operation, but it will mean that the scanning will be much slower.

tjhowse, 2020-11-01:
Righto. Give v0.3.3 a try. Add `scan_batching: 1` to your config and it will scan each register individually. I have a 50ms wait between each read, hopefully that won't overwhelm the modbus interface on the inverter.

rpvelloso, 2020-11-01:
My inverter has a default limit of 3 simultaneous connections. Does modbus4mqtt opens up simultaneous connections???

rpvelloso, 2020-11-01:
The wifi dongle, that is.

tjhowse, 2020-11-01:
Nope! It's a single-thread application. All comms is sequential.

plague-doctor, 2020-11-01:
hmmmm.... @tjhowse  soon you will hate me :-(
```
modbus4mqtt==0.3.3
SungrowModbusTcpClient==0.1.5
```
config.yaml:
```
ip: 172.16.50.116
port: 502
update_rate: 5
address_offset: 0
variant: sungrow
scan_batching: 1
registers:
  - pub_topic: "seconds"
    address: 5004
    pub_only_on_change: false
    table: 'holding'
  - pub_topic: "5016"
    address: 5016
  - pub_topic: "5002"
    address: 5002
  - pub_topic: "5007"
    address: 5007
  - pub_topic: "5012"
    address: 5112
  - pub_topic: "5030"
    address: 5030
  - pub_topic: "5035"
    address: 5035
  - pub_topic: "5018"
    address: 5018
  - pub_topic: "5021"
    address: 5021
  - pub_topic: "5143"
    address: 5143
  - pub_topic: "5037"
    address: 5037
  - pub_topic: "5010"
    address: 5010
  - pub_topic: "5011"
    address: 5011
  - pub_topic: "5012"
    address: 5012
  - pub_topic: "5013"
    address: 5013
  - pub_topic: "5043"
    address: 5043
```
published:
```
test/modbus4mqtt hi
test/5002 14
test/5002 14
```
errors:
```
2020-11-01 15:04:06 INFO     Connected to MQTT.
2020-11-01 15:04:12 ERROR    Failed to poll modbus device, attempting to reconnect: 'ExceptionResponse' object has no attribute 'registers'
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus4mqtt.py", line 72, in poll
    self._mb.poll()
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 71, in poll
    values = self._scan_value_range(table, group, self._scan_batching)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 143, in _scan_value_range
    return result.registers
AttributeError: 'ExceptionResponse' object has no attribute 'registers'
2020-11-01 15:04:23 ERROR    Failed to poll modbus device, attempting to reconnect: 'ExceptionResponse' object has no attribute 'registers'
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus4mqtt.py", line 72, in poll
[... 184 lines cut ...]
2020-11-01 15:04:38 WARNING  Couldn't get value from register 5112 in table holding
2020-11-01 15:04:38 WARNING  Couldn't get value from register 5030 in table holding
2020-11-01 15:04:38 WARNING  Couldn't get value from register 5035 in table holding
2020-11-01 15:04:38 WARNING  Couldn't get value from register 5018 in table holding
2020-11-01 15:04:38 WARNING  Couldn't get value from register 5021 in table holding
2020-11-01 15:04:38 WARNING  Couldn't get value from register 5143 in table holding
2020-11-01 15:04:38 WARNING  Couldn't get value from register 5037 in table holding
2020-11-01 15:04:38 WARNING  Couldn't get value from register 5010 in table holding
2020-11-01 15:04:38 WARNING  Couldn't get value from register 5011 in table holding
2020-11-01 15:04:38 WARNING  Couldn't get value from register 5012 in table holding
2020-11-01 15:04:38 WARNING  Couldn't get value from register 5013 in table holding
2020-11-01 15:04:38 WARNING  Couldn't get value from register 5043 in table holding
2020-11-01 15:04:43 ERROR    Unable to decode response Modbus Error: Unknown response 0
2020-11-01 15:04:43 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/transaction.py", line 190, in execute
    self.client.framer.processIncomingPacket(response,
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request

[...] and so on...
```

tjhowse, 2020-11-01:
Not at all! Have you got anything else talking to your inverter at the same time? A web interface open? Another modbus connection? It seems like the inverter can handle certain combinations of requests but can quickly be overwhelmed.

I'm also wondering if whether it having a connection to the web could also be a factor. Can your inverter access the internet? Is it reporting data to iSolarCloud at the moment? I've disabled that connection with a rule in my router. You can also give the inverter a bad gateway IP address to prevent it routing to the internet.

rpvelloso, 2020-11-01:
In the manual these are "input" registers, not "holding". Maybe this is it? I'm reading them all as input registers and it is working just fine.

tjhowse, 2020-11-01:
If you don't specify the table type it defaults to 'holding', so all of the registers in the example that @plague-doctor just posted will be scanned as holding register tables.

plague-doctor, 2020-11-01:
Ok, some more tests. 
I have gone through the `Modbus RS485 RTU Protocol.pdf` and put more registers in. Now my test looks like this:
```
ip: 172.16.50.116
# ip: 192.168.1.89
port: 502
update_rate: 5
address_offset: 0
variant: sungrow
scan_batching: 100
registers:
  - pub_topic: "seconds"
    address: 5004
    pub_only_on_change: false
    table: 'holding'
  - pub_topic: "5000"
    address: 5000
  - pub_topic: "5001"
    address: 5001
  - pub_topic: "5002"
    address: 5002
  - pub_topic: "5003"
    address: 5003
  - pub_topic: "5004"
    address: 5004
  - pub_topic: "5005"
    address: 5005
  - pub_topic: "5006"
    address: 5006
  - pub_topic: "5007"
    address: 5007
  - pub_topic: "5008"
    address: 5008
  - pub_topic: "5009"
    address: 5009
  - pub_topic: "5010"
    address: 5010
  - pub_topic: "5011"
    address: 5011
  - pub_topic: "5012"
    address: 5012
  - pub_topic: "5013"
    address: 5013
  - pub_topic: "5014"
    address: 5014
  - pub_topic: "5017"
    address: 5017
  - pub_topic: "5018"
    address: 5018
  - pub_topic: "5113"
    address: 5113
  - pub_topic: "5128"
    address: 5128
  - pub_topic: "5129"
    address: 5129
  - pub_topic: "5031"
    address: 5031
  - pub_topic: "5032"
    address: 5032
```
A lot less errors are thrown (still occasional hick-ups  , though):
```
2020-11-01 15:53:32 INFO     Connected to MQTT.
2020-11-01 15:56:47 ERROR    Failed to read holding table registers from 5000 to 5100
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 71, in poll
    values = self._scan_value_range(table, group, self._scan_batching)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 142, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read holding table registers from 5000 to 5100
2020-11-01 15:56:48 ERROR    Unable to decode response Modbus Error: Unknown response 0
2020-11-01 15:56:48 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/transaction.py", line 190, in execute
    self.client.framer.processIncomingPacket(response,
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2020-11-01 15:56:48 ERROR    Failed to read holding table registers from 5100 to 5200
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 71, in poll
    values = self._scan_value_range(table, group, self._scan_batching)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 142, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read holding table registers from 5100 to 5200
2020-11-01 15:57:00 ERROR    Failed to read holding table registers from 5000 to 5100
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 71, in poll
    values = self._scan_value_range(table, group, self._scan_batching)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 142, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read holding table registers from 5000 to 5100
2020-11-01 15:57:00 ERROR    Unable to decode response Modbus Error: Unknown response 0
2020-11-01 15:57:00 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/transaction.py", line 190, in execute
    self.client.framer.processIncomingPacket(response,
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2020-11-01 15:57:00 ERROR    Failed to read holding table registers from 5100 to 5200
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 71, in poll
    values = self._scan_value_range(table, group, self._scan_batching)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 142, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read holding table registers from 5100 to 5200

```
But finally I can see some outputs:
```
modbus4mqtt = hi
seconds = 56
5000 = 11
5001 = 1
5002 = 14
5003 = 55
5004 = 56
5005 = 207
5006 = 170
5007 = 1000
5008 = 255
5009 = 1000
5010 = 65535
5011 = 65535
5012 = 65535
5013 = 65535
5014 = 65535
5017 = 65535
5018 = 1000
5113 = 42353
5128 = 65535
5129 = 65535
5031 = 65535
5032 = 65535
```
I have no idea how to interpret the output... Especially registers 5003 and 5004...
Let me share a screenshot with you: https://seafile.cryptohell.net/d/94398451f9b0492e897b/

rpvelloso, 2020-11-01:
Can you change the registers from "holding" to "input" and see what happens?

plague-doctor, 2020-11-01:
all of them?

rpvelloso, 2020-11-01:
All that are listed in the PDF I sent as "input". There are two tables in there: input and holding.

tjhowse, 2020-11-01:
Note the table field is register-specific. You'll need an entry per register that you want to read from the input table.

The process of determining what a given register means, without a memory map from the manufacturer, is one of educated guessing. 😢 I'd say the ones that are 65535 are unused. 5002 is hours in 24-hour time, 5003 is minutes (Hey, fellow UTC+10 😉 ), 5004 is seconds.

plague-doctor, 2020-11-01:
https://seafile.cryptohell.net/d/94398451f9b0492e897b/files/?p=%2FScreenshot_1.png
Now it is interesting :-)
But what is what....?

rpvelloso, 2020-11-01:
The errors ceased?

plague-doctor, 2020-11-01:
Not that good yet :-)
```
2020-11-01 16:15:07 INFO     Connected to MQTT.
2020-11-01 16:18:54 ERROR    Failed to read input table registers from 5000 to 5100
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 71, in poll
    values = self._scan_value_range(table, group, self._scan_batching)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 142, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read input table registers from 5000 to 5100
2020-11-01 16:18:55 ERROR    Unable to decode response Modbus Error: Unknown response 0
2020-11-01 16:18:55 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/transaction.py", line 190, in execute
    self.client.framer.processIncomingPacket(response,
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2020-11-01 16:18:55 ERROR    Failed to read input table registers from 5100 to 5200
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 71, in poll
    values = self._scan_value_range(table, group, self._scan_batching)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 142, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read input table registers from 5100 to 5200
2020-11-01 16:19:01 ERROR    Failed to read input table registers from 5000 to 5100
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 71, in poll
    values = self._scan_value_range(table, group, self._scan_batching)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 142, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read input table registers from 5000 to 5100
2020-11-01 16:19:01 ERROR    Unable to decode response Modbus Error: Unknown response 0
2020-11-01 16:19:01 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/transaction.py", line 190, in execute
    self.client.framer.processIncomingPacket(response,
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2020-11-01 16:19:01 ERROR    Failed to read input table registers from 5100 to 5200
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 71, in poll
    values = self._scan_value_range(table, group, self._scan_batching)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 142, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read input table registers from 5100 to 5200
2020-11-01 16:19:08 ERROR    Failed to read input table registers from 5000 to 5100
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 71, in poll
    values = self._scan_value_range(table, group, self._scan_batching)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 142, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read input table registers from 5000 to 5100
2020-11-01 16:19:09 ERROR    Unable to decode response Modbus Error: Unknown response 0
2020-11-01 16:19:09 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/transaction.py", line 190, in execute
    self.client.framer.processIncomingPacket(response,
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
[... cut ...]
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 71, in poll
    values = self._scan_value_range(table, group, self._scan_batching)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 142, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read input table registers from 5100 to 5200
```

tjhowse, 2020-11-01:
I'm afraid those screenshot links are timing out for me. Are you able to attach them to your post in this issue instead?

plague-doctor, 2020-11-01:
Sorry. Of course.

[image]

plague-doctor, 2020-11-01:
I am getting closer to interpret those graphs. Or I am delusional from staring at them...
[image]
5002 - Today Yield (kWh)
5003 - Total Yield (kWh)
5016 - Actual power generated (W)
5010 - MPPT1 Voltage (0.1 V)
5011 - MPPT1 Amperage (0.1 A)
5012 - MPPT2 Voltage (0.1 V)
5013 - MPPT2 Amperage (0.1 A)
Not sure what are others... Temperature, etc...

plague-doctor, 2020-11-01:
still errors out from time to time:
```
2020-11-01 18:01:44 ERROR    index out of range
2020-11-01 18:01:44 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/transaction.py", line 190, in execute
    self.client.framer.processIncomingPacket(response,
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2020-11-01 18:01:44 ERROR    Failed to read input table registers from 5100 to 5200
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 71, in poll
    values = self._scan_value_range(table, group, self._scan_batching)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 142, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read input table registers from 5100 to 5200
2020-11-01 18:07:05 ERROR    Failed to read input table registers from 5000 to 5100
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 71, in poll
    values = self._scan_value_range(table, group, self._scan_batching)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 142, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read input table registers from 5000 to 5100
2020-11-01 18:07:06 ERROR    index out of range
2020-11-01 18:07:06 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/transaction.py", line 190, in execute
    self.client.framer.processIncomingPacket(response,
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2020-11-01 18:07:06 ERROR    Failed to read input table registers from 5100 to 5200
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 71, in poll
    values = self._scan_value_range(table, group, self._scan_batching)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 142, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read input table registers from 5100 to 5200
2020-11-01 18:07:13 ERROR    Failed to read input table registers from 5000 to 5100
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 71, in poll
    values = self._scan_value_range(table, group, self._scan_batching)
  File "/home/abadonna/.local/lib/python3.8/site-packages/modbus4mqtt/modbus_interface.py", line 142, in _scan_value_range
    raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
ValueError: Failed to read input table registers from 5000 to 5100
2020-11-01 18:07:13 ERROR    index out of range
2020-11-01 18:07:13 ERROR    Modbus Error: [Input/Output] Unable to decode request
Traceback (most recent call last):
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/transaction.py", line 190, in execute
    self.client.framer.processIncomingPacket(response,
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
    self._process(callback, error=True)
  File "/home/abadonna/.local/lib/python3.8/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
    raise ModbusIOException("Unable to decode request")
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
```

tjhowse, 2020-11-01:
That's to be expected. My inverter occasionally throws random errors, but overall it gets the job done.

tjhowse, 2020-11-05:
Closing this issue as the root cause is resolved, even if there are still other issues to handle.

closed 2020-11-05 by tjhowse

entropical, 2020-11-20:
Hey @plague-doctor, do you have a final .yaml file?

I've successfully got some numbers, using your sample configs above:

test/seconds 44
test/5004 44
test/seconds 50
test/5004 50
test/seconds 57
test/5004 57
test/seconds 3
test/5003 16
test/5004 3
test/seconds 9

tjhowse, 2020-11-21:
@entropical We've started up a new issue here: https://github.com/tjhowse/modbus4mqtt/issues/16

Can you please post your full inverter model number and firmware version in the new issue? I'm wondering if there might be something wrong with @plague-doctor's specific setup in some way, but I'm really not sure. If we have another point of comparison that is the same in every important respect we should have a better chance of pinning down the problem.

PR #15 (2020-11-01): Custom scan batching
Add a configurable setting to allow for different batched modbus reads. This could help resolve #13 .