tjhowse/modbus4mqtt #16: modbus4mqtt errors out on SG5K-D

plague-doctor, 2020-11-05:
This is a different issue to https://github.com/tjhowse/modbus4mqtt/issues/13 but related.

The module constantly throws errors:

```
solar2mqtt | 2020-11-05 20:21:32 ERROR    Failed to read input table registers from 5000 to 5100
solar2mqtt | Traceback (most recent call last):
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 72, in poll
solar2mqtt |     values = self._scan_value_range(table, group, self._scan_batching)
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 143, in _scan_value_range
solar2mqtt |     raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
solar2mqtt | ValueError: Failed to read input table registers from 5000 to 5100
solar2mqtt | 2020-11-05 20:21:38 ERROR    Unable to decode response Modbus Error: Unknown response 0
solar2mqtt | 2020-11-05 20:21:38 ERROR    Modbus Error: [Input/Output] Unable to decode request
solar2mqtt | Traceback (most recent call last):
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/pymodbus/transaction.py", line 190, in execute
solar2mqtt |     self.client.framer.processIncomingPacket(response,
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
solar2mqtt |     self._process(callback, error=True)
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
solar2mqtt |     raise ModbusIOException("Unable to decode request")
solar2mqtt | pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
solar2mqtt | 2020-11-05 20:21:38 ERROR    Failed to read input table registers from 5000 to 5100
solar2mqtt | Traceback (most recent call last):
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 72, in poll
solar2mqtt |     values = self._scan_value_range(table, group, self._scan_batching)
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 143, in _scan_value_range
solar2mqtt |     raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
solar2mqtt | ValueError: Failed to read input table registers from 5000 to 5100
solar2mqtt | 2020-11-05 20:21:44 ERROR    Failed to read input table registers from 5000 to 5100
solar2mqtt | Traceback (most recent call last):
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 72, in poll
solar2mqtt |     values = self._scan_value_range(table, group, self._scan_batching)
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 143, in _scan_value_range
solar2mqtt |     raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
solar2mqtt | ValueError: Failed to read input table registers from 5000 to 5100
solar2mqtt | 2020-11-05 20:21:50 ERROR    Failed to read input table registers from 5000 to 5100
solar2mqtt | Traceback (most recent call last):
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 72, in poll
solar2mqtt |     values = self._scan_value_range(table, group, self._scan_batching)
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 143, in _scan_value_range
solar2mqtt |     raise ValueError("Failed to read {} table registers from {} to {}".format(table, start, start+count))
solar2mqtt | ValueError: Failed to read input table registers from 5000 to 5100
```
and finally stops reporting (most of the cases at night). Memory leak?

My config SG5K-D.yaml:
```
ip: 172.16.50.116
port: 502
update_rate: 5
address_offset: 0
variant: sungrow
scan_batching: 100
registers:
  - pub_topic: "5016"  # Active power generated
    address: 5016
    table: 'input'
  - pub_topic: "5002"  # Daily yield
    address: 5002
    table: 'input'
  - pub_topic: "5003"  # Total yield
    address: 5003
    table: 'input'
```

tjhowse, 2020-11-06:
Hm, could you try a `scan_batching: 1` and see what changes?

tjhowse, 2020-11-06:
Is there anything else talking to the inverter? The wifi interface can only handle a few connections at once, and this includes the connection to iSolarCloud.

plague-doctor, 2020-11-06:
there is only iSolarCloud talking.

tjhowse, 2020-11-06:
When I was using wifi I had one dropout every half an hour, all modbus comms was interrupted, and then it came back again. I switched to an ethernet connection and it's been solid since then. The SH5k-20 has an internal ethernet port.

Note that the RJ45 socket the wifi dongle uses is not an ethernet port. It's a serial RS-485 that just happens to use the same connector as ethernet.

plague-doctor, 2020-11-06:
changing `scan_batching` to 1 did not help. It is still failing quite a lot...

```
solar2mqtt | 2020-11-06 12:42:19 ERROR    index out of range
solar2mqtt | 2020-11-06 12:42:19 ERROR    Modbus Error: [Input/Output] Unable to decode request
solar2mqtt | 2020-11-06 12:47:55 ERROR    Failed to read input table registers from 5002 to 5003
solar2mqtt | 2020-11-06 12:47:55 ERROR    index out of range
solar2mqtt | 2020-11-06 12:47:55 ERROR    Modbus Error: [Input/Output] Unable to decode request
solar2mqtt | 2020-11-06 12:47:55 ERROR    Failed to read input table registers from 5003 to 5004
solar2mqtt | 2020-11-06 12:48:02 ERROR    Failed to read input table registers from 5002 to 5003
solar2mqtt | 2020-11-06 12:48:02 ERROR    index out of range
solar2mqtt | 2020-11-06 12:48:02 ERROR    Modbus Error: [Input/Output] Unable to decode request
solar2mqtt | 2020-11-06 12:48:02 ERROR    Failed to read input table registers from 5003 to 5004
solar2mqtt | 2020-11-06 12:48:09 ERROR    Failed to read input table registers from 5002 to 5003
solar2mqtt | 2020-11-06 12:48:09 ERROR    index out of range
solar2mqtt | 2020-11-06 12:48:09 ERROR    Modbus Error: [Input/Output] Unable to decode request
solar2mqtt | 2020-11-06 12:48:09 ERROR    Failed to read input table registers from 5003 to 5004
solar2mqtt | 2020-11-06 12:53:38 ERROR    Failed to read input table registers from 5002 to 5003
solar2mqtt | 2020-11-06 12:53:39 ERROR    tuple index out of range
solar2mqtt | 2020-11-06 12:53:39 ERROR    Modbus Error: [Input/Output] Unable to decode request
solar2mqtt | 2020-11-06 12:53:39 ERROR    Failed to read input table registers from 5003 to 5004
solar2mqtt | 2020-11-06 12:53:45 ERROR    Failed to read input table registers from 5002 to 5003
solar2mqtt | 2020-11-06 12:53:46 ERROR    tuple index out of range
solar2mqtt | 2020-11-06 12:53:46 ERROR    Modbus Error: [Input/Output] Unable to decode request
solar2mqtt | 2020-11-06 12:53:46 ERROR    Failed to read input table registers from 5003 to 5004
solar2mqtt | 2020-11-06 12:53:52 ERROR    Failed to read input table registers from 5002 to 5003
solar2mqtt | 2020-11-06 12:53:52 ERROR    tuple index out of range
solar2mqtt | 2020-11-06 12:53:52 ERROR    Modbus Error: [Input/Output] Unable to decode request
solar2mqtt | 2020-11-06 12:53:52 ERROR    Failed to read input table registers from 5003 to 5004
```

tjhowse, 2020-11-07:
Can you please try upgrading v0.3.4? I don't expect it will fix the issue but it may shed additional light on exactly what's going wrong in the internals thanks to some logging improvements.

Could you also try a scan_batching value of 10? Some example documentation I've found for communicating with Sungrow inverters uses a read size of 10, so maybe their dodgy implementation of modbus expects that exact value and gets upset otherwise?

plague-doctor, 2020-11-07:
Sure. Let's try to throw some lights:
```
solar2mqtt | 2020-11-07 18:20:26 ERROR    Failed to read 10 input table registers starting from 5010: Modbus Error: [Input/Output] Unable to decode request
solar2mqtt | Traceback (most recent call last):
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 143, in _scan_value_range
solar2mqtt |     return result.registers
solar2mqtt | AttributeError: 'ModbusIOException' object has no attribute 'registers'
solar2mqtt | 
solar2mqtt | During handling of the above exception, another exception occurred:
solar2mqtt | 
solar2mqtt | Traceback (most recent call last):
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 72, in poll
solar2mqtt |     values = self._scan_value_range(table, group, self._scan_batching)
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 146, in _scan_value_range
solar2mqtt |     raise ValueError("Failed to read {} {} table registers starting from {}: {}".format(count, table, start, result))
solar2mqtt | ValueError: Failed to read 10 input table registers starting from 5010: Modbus Error: [Input/Output] Unable to decode request
solar2mqtt | 2020-11-07 18:31:07 ERROR    Failed to read 10 input table registers starting from 5000: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)
solar2mqtt | Traceback (most recent call last):
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 143, in _scan_value_range
solar2mqtt |     return result.registers
solar2mqtt | AttributeError: 'ModbusIOException' object has no attribute 'registers'
solar2mqtt | 
solar2mqtt | During handling of the above exception, another exception occurred:
solar2mqtt | 
solar2mqtt | Traceback (most recent call last):
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 72, in poll
solar2mqtt |     values = self._scan_value_range(table, group, self._scan_batching)
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 146, in _scan_value_range
solar2mqtt |     raise ValueError("Failed to read {} {} table registers starting from {}: {}".format(count, table, start, result))
solar2mqtt | ValueError: Failed to read 10 input table registers starting from 5000: Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 8 bytes (0 received)
solar2mqtt | 2020-11-07 18:31:08 ERROR    Unable to decode response Modbus Error: Unknown response 0
solar2mqtt | 2020-11-07 18:31:08 ERROR    Modbus Error: [Input/Output] Unable to decode request
solar2mqtt | Traceback (most recent call last):
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/pymodbus/transaction.py", line 190, in execute
solar2mqtt |     self.client.framer.processIncomingPacket(response,
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 165, in processIncomingPacket
solar2mqtt |     self._process(callback, error=True)
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/pymodbus/framer/socket_framer.py", line 175, in _process
solar2mqtt |     raise ModbusIOException("Unable to decode request")
solar2mqtt | pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
solar2mqtt | 2020-11-07 18:31:08 ERROR    Failed to read 10 input table registers starting from 5010: Modbus Error: [Input/Output] Unable to decode request
solar2mqtt | Traceback (most recent call last):
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 143, in _scan_value_range
solar2mqtt |     return result.registers
solar2mqtt | AttributeError: 'ModbusIOException' object has no attribute 'registers'
solar2mqtt | 
solar2mqtt | During handling of the above exception, another exception occurred:
solar2mqtt | 
solar2mqtt | Traceback (most recent call last):
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 72, in poll
solar2mqtt |     values = self._scan_value_range(table, group, self._scan_batching)
solar2mqtt |   File "/usr/local/lib/python3.9/site-packages/solar2mqtt/modbus_interface.py", line 146, in _scan_value_range
solar2mqtt |     raise ValueError("Failed to read {} {} table registers starting from {}: {}".format(count, table, start, result))
solar2mqtt | ValueError: Failed to read 10 input table registers starting from 5010: Modbus Error: [Input/Output] Unable to decode request
```

entropical, 2020-11-21:
My device is an SG5K-D. How do I get the firmware version? From the LCD?

tjhowse, 2020-11-21:
Yep! I'm not sure if the SG5k-D LCD interface is the same as the SH5k-20, but on mine the firmware version and serial number are available at the bottom of the "Run info" menu pages. I pressed ENT (menu), ENT (run info), then UP (device info).

entropical, 2020-11-21:
With the aid of the manual, I worked out how to drive the LCD screen.

My firmware version is: CrysG2_V31_FW_V22

tjhowse, 2020-11-21:
OK.
@plague-doctor reported M_WIFI_RAK475_V31_V01_Q as their firmware, but that's the firmware version of the wifi dongle. I have SH5k-V13_FW_V013 on my LCD and M_WIFI_RAK475_V25_V01_C on my (disconnected) dongle.

plague-doctor, 2020-11-29:
I have missed this somehow... The firmware version of my device is: **CrysG2_V31_FW-V22** (from LCD display).

rpvelloso, 2020-12-18:
They've added a new option to the app: live data. But the app says my model doesn't have it, can you believe it? I was getting live data, by myself, until yesterday :|

entropical, 2020-12-19:
Last night I was working on getting HomeAssistant to read directly from modbus. I was making some progress, and then it stopped. I traced backwards and found port 502 was closed. 

I was worried that I'd caused the inverter to crash from my polling, so was actually relieved to see your post!

Same thing in my app too, no live data for me.

tjhowse, 2020-12-19:
This is dire news. Very disappointed in Sungrow. Does anyone running the new firmware have access to the web interface? Is it possible to re-enable the modbus interface in there? Is it possible to downgrade the firmware via the iSolarHome smartphone app?

rpvelloso, 2020-12-21:
Well, this is very weird. Today my inverter came back to life. But still, all doors are closed. BUT my home assistant was connected to it and working flawlessly. I think they shut port 80 for good and 502 is shut after ONE connection is established. @entropical  can you, please, check if this is also happening on your equipment? Thanks.

tjhowse, 2020-12-22:
Hm. I speculate this might be sungrow's fix for unreliable modbus performance when multiple clients are talking to the same host. The wifi dongle talks to the inverter via a modbus-485 connection. If the dongle can't safely multiplex multiple parallel requests from modbus-tcp to serial requests over modbus-485 then it makes sense to only allow one modbus-tcp connection to the dongle.

rpvelloso, 2020-12-22:
Yeah, that's what I thought too.

rpvelloso, 2020-12-22:
This might be a problem depending on how pymodbus (or the app using it) handles tcp comms. You no longer can spawn several connections in parallel. Today my integration with home assistant was working, but the other day it was not. Most likely the connection was already in use and the dongle shut the modbus port.

tjhowse, 2020-12-22:
modbus4mqtt only opens a single modbus connection, I'm not sure about the
HASS modbus implementation.

plague-doctor, 2020-12-23:
Oh, it is good I have blocked SG5K-D on the router... 
By the way, is there a way to get info about how much I feed back to the grid?

rpvelloso, 2020-12-23:
@plague-doctor on my inverter (SG4K) is:
```      - name: total_active_power
        hub: sungrow_sg4k
        slave: 1
        count: 2
        register: 5030
        reverse_order: true
        register_type: input
        unit_of_measurement: W
```

But this only reports the total injected power, some of it is used right away and what is left is injected back to the grid. I think you need an specific installation in order to know exactly how much you inject (or just trust the electric company measurements :D).

plague-doctor, 2020-12-23:
If only I could have the right mapping for SG5K-D...

rpvelloso, 2020-12-23:
I have a file sungrow sent me, it is supposed to work with all residential models. Its attached. 
It doesn't  work for you?
[Uploading TI_20190704_Communication Protocol for Residential Single-phase Grid-Connected Inverters_V10_EN.pdf…]()

plague-doctor, 2020-12-23:
Could you please share this file?

rpvelloso, 2020-12-23:
https://github.com/tjhowse/modbus4mqtt/files/5732710/TI_20190704_Communication.Protocol.for.Residential.Single-phase.Grid-Connected.Inverters_V10_EN.pdf

plague-doctor, 2020-12-23:
Thank you kindly.
Could you please tell me how did you figure out that `5030` is what you say it is?
What I can see is:
```
23  Reserved   5023–5030
```

rpvelloso, 2020-12-23:
It is 'minus one' says in the document
Total active power 5031–5032 U32 W

Note:
Communication address = protocol address–1. If data of address 5000 is to be inquired, the corresponding
sending address data is 4999 (0x1387);

Don't ask my why...

plague-doctor, 2020-12-23:
Before I will fund a marble statue for you, can you double check this:
```
5002 - Daily yield (0.1 kWh)
5003 - Total yield (kWh)
5016 - Currently producing power (W)
5030 - Currently using (W)
Currently feeding to grid: 5016 - 5030 (W) ?
```
Is this right?

rpvelloso, 2020-12-23:
I think those are the same registers I'm using too, looks about right.
I like my statues about 5m tall at least :D

rpvelloso, 2020-12-23:
5016 is DC Power output from PV modules before AC conversion.
5030 is AC Power output form inverter to mains.
A validated this registers against sungrow's phone app.

plague-doctor, 2020-12-23:
So 5030 is what I get from the solar after AC conversion. 
In that case I do not know how much I use and/or feed back to grid. What do you use to figure the consumption out?

rpvelloso, 2020-12-23:
What is left is measure by the electric company's meter.

plague-doctor, 2020-12-23:
Ok. Thanks a lot. The statue will be 4.5m but green marble ;-)

tjhowse, 2020-12-27:
If either of you has a YAML file for the SG5k-D I'd be very keen to include it in this repo. Feel free to raise a pull request or attach it to this issue.

plague-doctor, 2020-12-27:
Here you are, but this is a very simple one... 

[SG5K-D.zip](https://github.com/tjhowse/modbus4mqtt/files/5744098/SG5K-D.zip)

rpvelloso, 2021-01-08:
web and modbus interfaces are backup to normal. \o/

tjhowse, 2021-01-09:
Interesting! Was there a dongle or inverter firmware update recently? Or did something else change?

rpvelloso, 2021-01-10:
I can't perceive any changes in the web interface. They probably updated the FW again or rolled back.