tjhowse/modbus4mqtt #21: Modbus to sungrow inverter connection ideas

jmkelly, 2020-12-30:
I've noticed that when I initially connect modbus4mqtt on my SG-4K-D, I get a readout for all the registers, and unless I've got a register in there that is wrong, no errors.  As it keeps running though, I'm only getting readouts on a few registers that regularly update and lots of exceptions on less frequently changing registers.

I was wondering if this could be something to do with a single tcp connection staying alive for the whole time that modbus5mqtt is running?  I've looked at the code, and I'm not all that familiar with python, but am I correct in assuming that once a tcp connection is made to the inverter, it is kept open?  If this is the case, then I was wondering what your thoughts on tearing down the connection and then re establishing it each tcp call to the inverter?

jmkelly, 2020-12-30:
I did some more reading.  From what I can tell, the connection is being established, then isn't explicitly closed.

In the SunGrowModbus library at https://github.com/rpvelloso/Sungrow-Modbus/blob/59130209e5e77dd105cd17c012c6a83f3c864bb7/SungrowModbusTcpClient/SungrowModbusTcpClient.py#L48 it closes the existing connection when a new call to connect is made.  

So is it right that the connection once made, will hold open until it polls again?  

I'd be interested in spiking out and testing connect, read data and close the connection explicitly, but I'm struggling to find exactly where the open / read / close would sit.  And also what implications of doing this may have....

tjhowse, 2020-12-30:
Correct - We start up a TCP connection when the service starts and it remains open as long as it can. If you wanted to experiment with opening and closing the connection with each poll, a good place for that would be in modbus_interface.poll()

https://github.com/tjhowse/modbus4mqtt/blob/master/modbus4mqtt/modbus_interface.py#L62

And then remove the connect code here: https://github.com/tjhowse/modbus4mqtt/blob/master/modbus4mqtt/modbus4mqtt.py#L41

However this isn't the intended use of a modbus interface. It's expecting to have a long-running connections open. Reconnecting on every poll would result in a lot of overhead. I would love to understand the real reason for the errors rather than implementing a workaround like this. I've connected directly to my SH5k-20 via ethernet rather than using the wifi-to-rs485 solarinfo dongle. I still get errors, but not enough to affect operation. I suspect part of the reason for my errors is the fact I'm running both pvstats and modbus4mqtt, and they're giving the inverter too much to think about.

Maybe there's some kind of keepalive signal the Sungrow inverter is expecting? Like a heartbeat signal? I'd like to look at a packet capture of a "proper" modbus TCP connection to see whether the interface is expecting something like that.

I strongly suspect that the Sungrow implementation of modbus is only "inspired" by the modbus spec, not actually modbus-compliant. There may be some interface settings that can be tweaked on our side to make the connections happier.

I have a task on the backlog to report multiple values in one message to MQTT via a JSON message: #9 Once this is implemented I can retire pvstats from my system and work on getting the error rate down. However my system is different to most in two ways: A) I'm not using wifi and, B) I've blocked the inverter from talking to the internet.

jmkelly, 2020-12-31:
thanks for the information, I'm a total noob to modbus, so its good to get more perspective.  Any suggestions on how to do the packet capture?  I'm running over wifi and connected to internet, so I'm thinking i could pick up the outgoing traffic to iSolarCloud if I can intercept the traffic going via my router (google wifi) somehow, but again would be interested in any ideas.

tjhowse, 2021-01-01:
When I've been sniffing the connection to my inverter I used a [packet squirrel](https://docs.hak5.org/hc/en-us/categories/360000982574-Packet-Squirrel), however they only work on ethernet connections - not wifi.

In any case the link from domestic inverters to the sungrow cloud services is a pseudo-encrypted MQTT link, not modbus. We'd need a reference implementation of a sungrow-endorsed modbus interface to one of their devices to classify exactly what connection parameters keep the connection happy. At the moment modbus4mqtt assumes that it doesn't need to do anything special to the connection to keep it connected. I know a few companies across the world are working on custom integrations to home inverters, but I'm not aware of anything specific to sungrow yet.

tjhowse, 2021-01-02:
I've just remembered I had this doc floating around. I have it referenced in another project, page 89 is most interesting: https://c.tjhowse.com/misc/SolarInfo%20Logger%20User%20Manual.pdf

jmkelly, 2021-01-03:
There’s some great info in there, but I can’t join the dots. What are you
thinking?

tjhowse, 2021-01-04:
Nothing specific - just adding some potentially useful context.

icypete, 2021-01-14:
I'm wondering if a middle ground solution would be to set a reconnect period, after which the connection is closed and reconnected. This would be independent of the polling period. For example we can poll every 5 seconds, but every (say) 10min the TCP connection is closed and re-established.