tjhowse/modbus4mqtt #26: Wrong address for total running time in SG5K-D.yaml

icypete, 2021-01-12:
In the example file SG5K-D.yaml the address for total running time is incorrect. The file has

<snip>
  - pub_topic: "total_yield"  #Total yield kWh
    address: 5003
    table: 'input'
  - pub_topic: "total_running_time"  #Total running time (h)
    address: 5003
    table: 'input'
<snip>

I think 5003 should be replaced by 5005 for total_running_time

icypete, 2021-01-13:
also missing "scale:" for most entries. i can update the file if it is helpful.

tjhowse, 2021-01-13:
Unfortunately I don't have an SG5k-D to test with, @jmkelly is there a chance you could see if address 5005 contains a constantly-increasing value on your system that could be a total_running_time?

Further, looking at these docs for the SH5k-20:

Holding registers:
[image]

Input registers:
[image]

These numbers include a +1 offset from how addresses are represented in modbus4mqtt, so assuming this memory map for the SH5k-20 is accurate for an SG5k-D, input table 5003-5004 is a uint32 for total energy output and holding table 5003 is system clock minutes? I'm not seeing anything in this dock about a total_running_time value though.

icypete, 2021-01-14:
ah. the guide i was using has;

[image]

I have an SG8K-D inverter and the 5005 address does contain a value that increases by 1 each hour. i have been assuming the SG8K-D and SG5K-D use the same mappings though

icypete, 2021-01-14:
[SG8K-D.zip](https://github.com/tjhowse/modbus4mqtt/files/5813039/SG8K-D.zip)

Attached is a working sample config file for SG8K-D. It was created from the SG5K-D.yaml file but I correct an address that was wrong and added the correct "scale:" values that were missing.

Note: it includes an example of a 32 bit signed value that needs the code changes I attached to #6 

I have tested the config with my SG8K-D unit and all fields work.

jmkelly, 2021-01-15:
oops, my bad, I had just a typo.  I've updated to total_running_time to 5005 on my SG5K-D, but now the sun is gone.

Once the changes are merged from #6 I'll give the new registers a crack (as I'm particularly interested in the charge back number) and happy to update the SG5K-D yaml.