tjhowse/modbus4mqtt #27: Cannot convert signed 32 bit values

icypete, 2021-01-12:
Some addresses like Energy Meter Power (address 5083-5084) are signed 32 bit values. There doesn't seem to be a way to read these values (type: int32 did not work). Also, the code looks to read all values as uint16 first and then try and convert them to an actual type so maybe 32 values can't be converted at the moment.

icypete, 2021-01-12:
oh, just saw that in modbus4mqtt.py there is to following line in validate_registers so I assume int32 is not supported yet. HAppy to help code this is needed.

        valid_types = ['uint16', 'int16']

tjhowse, 2021-01-12:
That would be a great feature to have! I've got an issue outstanding for multi-word register support here: #6 Let's centralise the discussion there.

closed 2021-01-12 by tjhowse

icypete, 2021-01-12:
Opps, sorry, neglected to check the open issues.