tjhowse/modbus4mqtt #36: Home Assistant compatibility

highground88, 2021-09-10:
Any chance this can be modified to be a valid Home Assistant repository (Or HAC)?

e.g: https://github.com/ttvt/ModbusTCP2MQTT

tjhowse, 2021-09-10:
Sounds quite doable. I'm adding this link here for my own reference for the requirements for a repo to work via HACS: https://hacs.xyz/docs/publish/start

tjhowse, 2021-10-17:
Hm, it looks like the best fit would be an "add-on" for HASS: https://developers.home-assistant.io/docs/add-ons/

highground88, 2021-10-17:
> Hm, it looks like the best fit would be an "add-on" for HASS, as opposed to an integration or plugin: https://developers.home-assistant.io/docs/add-ons/

Happy to test - I'm currently using "https://github.com/TenySmart/ModbusTCP2MQTT" as an integration through Supervisor custom repo which seems to be working mostly ok

highground88, 2021-10-17:
Sorry looks like some incompatibility still? Thanks for your efforts

HA custom
[image]

HACS custom
[image]