tjhowse/modbus4mqtt #45: Please add Float32 and float64 Support

Pubaluba, 2023-06-20:
Would be Nice to have the ability to publish the Float32/float64 valzes of Energy Meter (kWh/Power/Vorlage/.....). to mqtt Server.

Would even donate for this feature.

Thank you for your good work.

pki791, 2023-06-20:
You can use the https://github.com/a-s-z-home/modbus4mqtt fork.

Pubaluba, 2023-06-20:
How to Install?
Readme is the same (pip3 -install ....)

git clone....?

pki791, 2023-06-20:
Same as this. As it is a fork it has only some changes.

Pubaluba, 2023-06-20:
> Same as this. As it is a fork it has only some changes.

??
How to say pip3 tonuse the fork?

Please Help.
Thanks

tjhowse, 2023-06-26:
This is definitely something I'm keen on adding. I'm moving house currently, and have a fair bit of other work going on, so I'm a bit time-limited. @JPSteindlberger has done a large amount of the work involved in implementing this feature. I hope to build on their work to get this merged before too long.

Pubaluba, 2023-07-05:
Hi, please merge this fork with Serial and unit-ID support.

Gut says IT can bei merge automaticaly, but im Not a collobarator.
IT can easyly selected @pull requests accross forks.
It schould bei done in few minutes.

Thank you.

Pubaluba, 2023-07-13:
> Same as this. As it is a fork it has only some changes.

After replacing Files and instaling ccorp module:
errors occour :
So, how to use the fork?

root@ips:~# modbus4mqtt --config /root/mqtt/oelheizung.yaml
2023-07-13 22:34:25 INFO     Starting modbus4mqtt v0.6.1
Traceback (most recent call last):
  File "/usr/local/bin/modbus4mqtt", line 8, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.10/dist-packages/click/core.py", line 1157, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/click/core.py", line 1078, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.10/dist-packages/click/core.py", line 1434, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.10/dist-packages/click/core.py", line 783, in invoke
    return __callback(*args, **kwargs)
  File "/usr/local/lib/python3.10/dist-packages/modbus4mqtt/modbus4mqtt.py", line 418, in main
    i.connect()
  File "/usr/local/lib/python3.10/dist-packages/modbus4mqtt/modbus4mqtt.py", line 71, in connect                                                              self.connect_modbus()
  File "/usr/local/lib/python3.10/dist-packages/modbus4mqtt/modbus4mqtt.py", line 75, in connect_modbus                                                       self._mb = modbus_interface.modbus_interface(self.config['url'],
KeyError: 'url'
root@ips:~#

pki791, 2023-11-16:
Have You been able to run it?

Pubaluba, 2023-11-16:
https://github.com/Pubaluba/modbus4mqtt_rebuild/

i made my own fork, and added float and double Support.

pki791, 2023-11-16:
> https://github.com/Pubaluba/modbus4mqtt_rebuild/
> 
> i made my own fork, and added float and double Support.

I also have my own fork already, i modified the poll loop to be executed on time, in that case you will not need to run it from cron, python is starting very long, specially on low end hardware. Also fixed some other small things and mainly the "RetryOnEmpty" settings, you should fix that too, Your version will not retry if the slave does not answer for the first time.

https://github.com/pki791/modbus4mqtt

closed 2023-11-16 by Pubaluba