tjhowse/modbus4mqtt #49: Change unit to slave in modbus_interfaces

nestor-alonso, 2023-09-26:
Hello!

I've been using a modified version of modbus4mqtt (applied the modbus RTU patches and added  some custom debugging). When running on Debian 12 (Bookworm) the modbus client wasn´t answering at all, so after some low-level debugging,  The solution was to change in modbus_interfaces.py all ocurrences of unit=slave for slave=slave. 

It seems that Pymodbus completed the transition from unit to slave.

I apologize for not providing a proper patch.

Best regards!

nestor-alonso, 2023-09-26:
Yep.... Now I see that listed in the requirement.txt. Then (question for myself) how the version 3.5.2 was installed in that system?

pki791, 2023-09-26:
I think that should be updated in the requirements.txt, there are more changes needed for 3.x as some procedure names also changed.

tjhowse, 2025-10-30:
v1.0.0 will support pymodbus >3, <4.

closed 2025-10-30 by tjhowse