tjhowse/modbus4mqtt #52: Request: add support for unit ID for Huawei inverters

TeslaV1, 2024-02-17:
Huawei inverters use a dongle as bridge to accept modbus tcp requests and map them to modbus rtu. With this more than one inverter can be connected to the dongle.

When wanting to read registers from one or the other inverter, the unit ID is needed to identify the correct inverter.

Please add support for that, many thanks in advance.

TeslaV1, 2024-02-17:
Dammit, is already implemented, sorry for that.

TeslaV1, 2024-02-17:
Sorry I have to reopen that.

This is my config:

ip: 192.168.0.157 #your inverter ip address
port: 502
update_rate: 15
device_address: 2 (tested with different addresses)
address_offset: 0
variant: tcp (tested with rtu-over-tcp)
word_order: highlow
scan_batching: 1
registers:
  - pub_topic: "modbus4mqtt/energy_yield"
    address: 32106
    table: 'holding'
    type: int32

Device Address is 2.
If I look at the packets in wireshark I see:
Unit Identifier 1

Unit Identifier does not change if I change it in the config.

Any tips?
Error is:
ValueError: Failed to read 1 holding table registers starting from 32107: Modbus Error: [Input/Output] No Response received from the remote unit/Unable to decode response

tjhowse, 2025-10-31:
It looks like this will be resolved in v1.0.0:

[image]

```
ip: 192.168.1.50
port: 5020
update_rate: 1
device_address: 2
registers:
  - pub_topic: "holding"
    address: 1
  - pub_topic: "input"
    table: "input"
    address: 1
  - pub_topic: "default"
    address: 2
```